cleanflight/blackbox-tools #1: Add support for decoding current sensor data

thenickdude, 2015-01-22:
Add support for decoding current sensor data in `blackbox_render` and `blackbox_decode`

thenickdude, 2015-01-26:
`blackbox_decode` support is in but not tested yet, `blackbox_render` support still to come.

fiveangle, 2015-01-31:
I came here to submit the same. Nice :)

-=dave

fiveangle, 2015-01-31:
Actually, it looks like this isn't the same: my thought is to replicate the virtual current sensor calculation from CF based on the BB data and report it in the decode/render tools. This would be independent of any actual current sensor data. For implementations where current sensor data is in the BB logs, it would also serve as a mechanism to hone the realtime virtual current sensor algorithm by comparison to the actual logs. 

Is that the scope you intended, or should I submit an new issue ?

-=dave

thenickdude, 2015-01-31:
Yeah that's exactly what I intend, there's no other way to do that.

fiveangle, 2015-01-31:
:+1: 

-=dave

digitalentity, 2015-03-30:
Hi Is there a timeframe for emulating virtual current sensor data in blackbox-render and blackbox-decode? @thenickdude, should I look into this myself, test it and open a PR?

thenickdude, 2015-03-30:
I'll implement that tonight @digitalentity!

thenickdude, 2015-03-30:
I've now implemented a simulated current meter and simulated cumulative energy meter in `blackbox_decode` with the `--simulate-current-meter` option in d11273780bee8d2442c38eff507badc095868943 (it's the CF algorithm, and it's tunable with --sim-current-meter-scale and --sim-current-meter-offset options). 

If you have a physical current meter, it also calculates cumulative energy usage in milliamp-hours for you using those ADC measurements.

digitalentity, 2015-03-30:
Thanks! Can this be done for blackbox_render as well?