cleanflight/blackbox-tools #3: add current PID configuration to blackbox data/render

nine-fpv, 2015-04-14:
would be very useful to have the current PID settings as a renderable item for the overlay. will be very helpful for those that use in flight adjustments to tune.

fedorcomander, 2015-06-22:
Hopefully will be merged as PR #7

thenickdude, 2015-06-24:
#7 has been merged so inflight adjustment events are now decoded by `blackbox_decode` into the .event file. They will also now appear in [the Web viewer for flight logs](https://github.com/cleanflight/blackbox-log-viewer). 

The `blackbox_render` tool doesn't yet support rendering of events so they're not appearing there yet.

Eventually the blackbox will be writing the entire system configuration to the log at arming time so we'll be able to inspect all settings.

closed 2023-04-19 by hydra