cleanflight/blackbox-tools #9: Bus error: 10

LaserLock, 2015-07-05:
i got this error on 2 different logs, the decode had no errors. 
other index on the same log worked.
why do i get it and how i prevent it in the future?

(i use a mac-mini with osx 10.10.4 and the tools 0.4.1)

thenickdude, 2015-07-06:
Please email me your log file ( n.sherlock@gmail.com ), or upload it somewhere where I can download it.

LaserLock, 2015-07-06:
sure ;)

http://rc-team-austria.com/uploads/LOG00006.TXT - index 2
http://rc-team-austria.com/uploads/LOG00009.TXT - index 1

thenickdude, 2015-07-06:
This looks to be a bug in the Cairo drawing library which is fixed in a newer version of Cairo. I'll get a new version of blackbox_render built that includes the fix and let you know!

thenickdude, 2015-07-09:
Fixed in fc67b11c7352e3b0599a16afc709ed49bc241a93, please grab this new release:

https://github.com/cleanflight/blackbox-tools/releases/tag/v0.4.1.1

closed 2015-07-09 by thenickdude