cleanflight/blackbox-tools #12: blackbox_render does not work with luxfloat

kaefert, 2015-08-09:
I just tried the luxfloat PID controller and the blackbox-log of that flight can be decoded just fine, but the rendering shows a unchanging PID table, unchanging command inputs and missing all the lines of motor outputs and gyro.

blackbox_decode works just fine and produces a csv document that looks all right.

kaefert, 2015-08-09:
oh... sorry, I simply tried again (a new flight also with luxfloat pid controller) and this time both decode and render tools worked. but it's still strange though that with my first luxfloat try decode worked but render delivered such a strange result although the decoded csv file looks like all the infos are there

thenickdude, 2015-08-14:
Please email me the log at n.sherlock@gmail.com

kaefert, 2015-08-17:
mail sent. hope you find time to take a look and maybe find the reason for the strange behaviour :)

thenickdude, 2015-08-19:
Ah I think the issue is that the timestamps are so large that they wrap around and end up looking negative. At least in the [web viewer](https://github.com/cleanflight/blackbox-log-viewer) this is the case:

[image]

It looks like your flight didn't start until 43 minutes after your flight controller was powered, is that accurate?

As an aside, your P and D are wayyy too high, which is causing those incredible oscillations on the motor commands (you can see that the gyro doesn't move at all, but the P and D lines are making huge overreactions).

kaefert, 2015-08-19:
ahh! yes that's possible. I often first power the flight controller (+gps module powered by the 3.3V output of my flight controller) with an usb cable to get a gps lock. Once that's done which sometimes takes a while (especially with that build where the gps antenna was looking down because of https://github.com/cleanflight/cleanflight/issues/1183 ) I connect the lipo before removing the usb cable to not loose the gps lock.

I guess it's possible I lost track of time reading some webpages or something and only started flying 43 minutes after first connecting the usb cable.

And yes I definitely did not yet get the hang of PID tuning... I'll need to get back to trying to learn that, at the moment my quad is pulled down and is in the progress of beeing rebuilt into a new frame that I designed and printed http://www.thingiverse.com/thing:974479

I think the PIDs in this flight where the defaults from the cleanflight configurator for the luxfloat PID-controller.. I've later tried to reduce the PIDs with the luxfloat controller but then in felt really heavy and sluggish.

I've mostly flown with the default PID-controller with default PIDs until now, which from the feeling of it was quite fine, though the blackbox overlay shows ugly oscilations on the motor outputs: https://www.youtube.com/watch?v=iRjbB9s6cwM

thenickdude, 2015-08-19:
Okay great, thanks! I'll see about fixing the timestamp display. It will be more difficult to display powerup times in excess of 71 minutes, but I should be able to get >35 minute recordings to display properly without any drama.

thenickdude, 2015-08-19:
This should now be fixed in [blackbox_render v0.4.2](https://github.com/cleanflight/blackbox-tools/releases) and the latest [web log viewer](https://github.com/cleanflight/blackbox-log-viewer), thanks for the log!

closed 2015-08-19 by thenickdude