cleanflight/blackbox-tools #17: Decode does not work with new betaflight .BFL file format

goatspoon, 2017-01-24:


thenickdude, 2017-01-24:
Really? My understanding was that Betaflight has kept the logs compatible. Do you have an example log file I can try?

goatspoon, 2017-01-24:
It does surprise me too. I managed to get a partial decode using --raw and by editing the file. But it still doesnt give a correct decode. https://www.dropbox.com/s/fufbfwpwighvhg2/LOG00004.BFL?dl=0

thenickdude, 2017-01-24:
Ah, it's due to this new predictor they added:

https://github.com/betaflight/betaflight/commit/70542eabd05319232470d1a1e2700c6083b1a96a

I'll add support for it.

goatspoon, 2017-01-24:
Very much appreciated.

thenickdude, 2017-01-24:
Please try out the new version 0.4.3, it should support your log now:

https://github.com/cleanflight/blackbox-tools/releases/tag/v0.4.3

closed 2017-01-24 by thenickdude

goatspoon, 2017-01-25:
Still getting "Unsupported field predictor 11"

thenickdude, 2017-01-25:
Ah I didn't rebuild the Windows binary! One second, I'll rebuild that and reupload.

thenickdude, 2017-01-25:
Please download the release again and give it another go!

goatspoon, 2017-01-25:
Perfect, thanks for your help. to EXCEL!