cleanflight/blackbox-tools #19: Deadlock while reading modern betaflight 3.2.2 file

Manawyrm, 2017-12-02:
Hi,

I have just upgraded my setup to betaflight 3.2.2, and now the blackbox-tools all seem to deadlock on my blackbox logs. 
These logs were created on the SD card in an SPRacing F3 NEO board, used to work nicely with cleanflight in the past. 

Even with the raw and debug-parameters the software just deadlocks and uses 100% CPU. 
blackbox_render, blackbox_decode and the Chromium-webapp all show this behaviour. 

https://screenshot.tbspace.de/xawdoshiczy.png
https://tbspace.de/content/downloads/LOG00008.BFL

Any ideas on what could be wrong with the new files? (i have 8 of these here, happens with all of them). 

Thanks,
Tobias

Manawyrm, 2017-12-02:
I just ran GDB on the file. I'm not a C expert at all, but it looks like the frameIndex is continually increasing beyond anything reasonable :-) 

https://screenshot.tbspace.de/mrfcdsahojk.png

Plasmatree, 2018-02-23:
I got the same problem but on 3.2.0 . Strangely a log I previously recorded on the same system doesn't show this behaviour. Both from a Matek f405 mini. Only difference is the logging frequency: 8khz works, 1.5khz log freezes chrome app and blackbox_decode.

closed 2023-04-19 by hydra