cleanflight/blackbox-tools #32: Poor performance on Rpi3 ubuntu server

LuisDiazUgena, 2019-10-30:
Hi,

I'm developing a bot that uses blackbox decode and the performance is really poor under this system. I understand that the HW used is not the best for this task, and using best hw will improve the way it works.

As i see on htop, the software is just using one core to perform the actions, I don't know if it will be possible to do something to improve runtime:

here is a capture of the htop command showing the cpu usage:

[image]

And the task itself:

[image]

as you can see, PID task 10052 is been running now for almost half an hour, when on a windows 10 system (it takes about 1 or 2 minutes top). 

Is anything I can do? I've downloaded and compiled the code on the raspberry pi 3. Log file:

This is de log that i'm using to test it:

[btfl_004.zip](https://github.com/cleanflight/blackbox-tools/files/3788328/btfl_004.zip)

Thanks.

thenickdude, 2019-10-30:
How the hell are you getting decode times so slow on Windows 10? It takes 2.15 seconds to decode the log you attached my machine.

LuisDiazUgena, 2019-10-30:
sorry, i miss spelled that. The whole process (bot reciving file, passing to plasmatree, decoding it, composing it and sending back) takes about that time.

thenickdude, 2019-10-30:
What is the actual runtime of blackbox_decode on the Raspberry Pi if you call it directly without using your wrapper software? There's no reason why the runtime should approach 20 seconds, let alone exceed it.

LuisDiazUgena, 2019-10-30:
ubuntu@ubuntu:~/bots/plasmatreebot/plasmatreeUnix/PID-Analyzer-master/blackbox-tools$ obj/blackbox_decode /home/ubuntu/bots/plasmatreebot/downloadDir/539011/btfl_004.bbl
Decoding log '/home/ubuntu/bots/plasmatreebot/downloadDir/539011/btfl_004.bbl' to '/home/ubuntu/bots/plasmatreebot/downloadDir/539011/btfl_004.01.csv'...
^\Quit (core dumped)

thenickdude, 2019-10-30:
Can you run it on your target under GDB and collect the stacktrace of the crash?

LuisDiazUgena, 2019-10-30:
can you explain how please?

thenickdude, 2019-10-30:
You should be able to run "gdb blackbox_decode" then "run yourlogfile.bbl". When it crashes it should print a stacktrace, or else you can enter "bt" to print one out. (You'll probably need to install a package to get gdb)

LuisDiazUgena, 2019-10-30:
Thanks so much. 

It's been runing for over an hour now and it doesn't seems to be working properly:

[image]

it does nothing:

[image]

I'm going to delete this and start over. Maybe something is wrong? I didn't see any errors while doing make

LuisDiazUgena, 2019-10-30:
it's seems that something was wrong with the make. It's working like a charm now!  takes lees than 10 seconds to do it (i was typing somewhere else and didn't notice exactly)

This is the output for make:

[image]

closed 2019-10-30 by LuisDiazUgena

mikeller, 2019-10-31:
@LuisDiazUgena: You might also want to look at https://github.com/betaflight/blackbox-tools - there are some recent changes / improvements in there that are not reflected in this repository.

LuisDiazUgena, 2019-11-02:
yeah, i know. I got their last version, thanks for sharing anyway!